LumaTeam/Luma3DS #849: Tales of the abyss Hard Crash

Rikimaruaxu, 2017-09-14:
Console New 2DS

SysNAND 11.5.0-38U

Boot9strap 1.2 
 Luma3DS 8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Game has been working great but leads to a crash when entering Mushroom Road
ErrDisp d900458b

**Steps to reproduce:**

1.Load up a save with mushroom road side quest unlocked

2.Enter Mushroom Road

[crash dump.zip](https://github.com/AuroraWright/Luma3DS/files/1303438/crash.dump.zip)

Rikimaruaxu, 2017-09-14:
And after reinstalling it wont even open. Other games will but the game this is about now crashes with a 
Error Type Generic
ProcessID 15
Process Name: Menu
Title ID:4003000008f02
Address: 0x00229350
Error Code: 0xd9004589

Doesn't provide a crash DMP

Hikari-chin, 2017-09-14:
Sounds like save data or game corruption.

closed 2017-09-14 by Hikari-chin

Rikimaruaxu, 2017-09-14:
Gonna try to reformat my SD card. which sucks since its a 64 gig with like 35 gigs of games on it..