LumaTeam/Luma3DS #850: Arm11 exception when turning off 3DS during setup

spcrypt2, 2017-09-15:
**System model:**
New 3DS
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

8.1.1 stable 

**Explanation of the issue:**

Luma3DS creates an exception, seemingly a kernel panic, when the system is force shutdown using the power button during the initial setup screen. 

**Steps to reproduce:**

1. Format System Memory
**System should restart after selecting 'OK'**

2. Wait for initial setup screen

3. Press power button until console shuts down

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1305052/crash_dump_00000000.zip)

Hikari-chin, 2017-09-15:
I'm pretty sure this isn't Luma's fault, iirc PTM really does crash if you do this, it just isn't caught on stock or CFWs without proper exception handlers.

closed 2017-09-15 by Hikari-chin

Margen67, 2017-09-15:
The same thing happens if you turn off the console with mSD management so I don't see why this could also be ignored.