LumaTeam/Luma3DS #851: I can't launch OCS

battletoad2002, 2017-09-15:
[image]

[Dump.zip](https://github.com/AuroraWright/Luma3DS/files/1307422/Dump.zip)
My problem is when i try to launch OCS i get a message that tells me that a exception occurred, i don´t how to fix this. I´ve been digging through froums for hours but i still found noone who had the same error as 
me. I would really appriciate if someone would help me! :^]

Here is a picture of the Error and a Dump.
[Dump.zip](https://github.com/AuroraWright/Luma3DS/files/1307426/Dump.zip)

Committee, 2017-09-15:
Any particular reason your launching OCD if you have Luma already?

Hikari-chin, 2017-09-16:
OCS is for acquiring CFW and attempts to use several exploits to gain the necessary permissions. Not a Luma issue.

closed 2017-09-16 by Hikari-chin