LumaTeam/Luma3DS #852: Rosalina homebrew incompatibility

PixelSergey, 2017-09-16:
As everyone probably knows, rosalina's homebrew is very incompatible with homebrew. It does not allow for target selection, breaking things like SVDT and sploit_installers, to name a few. It also takes over hblauncherloader (for some reason?) and restricts that.

ihaveamac, 2017-09-16:
title selection was intentionally left out since 3dsx homebrew running under rosalina should have access to every title directly already. it just requires the programs be updated for it. JKSM is one such thing that should work fine.

it taking over hblauncher_loader is mostly just based on it being the best title for rosalina. others could be used and the same things would apply. *hax won't work because of rosalina.

Hikari-chin, 2017-09-16:
Target title selection isn't necessary under Rosalina because of expanded permissions. It takes over hblauncher_loader because *hax doesn't work under Rosalina anyway.

closed 2017-09-16 by Hikari-chin