LumaTeam/Luma3DS #854: Wrong color format in VC screenshot

Flaviculpus, 2017-09-16:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.5.0 SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.2

**Luma3DS version:**
8.1.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (2)

Splash: (on)

PIN lock: (off)

New 3DS CPU: (off)

--

Autoboot EmuNAND: (off)

Use EmuNAND FIRM if booting with R: (off)

Enable loading external FIRMs and modules: (off)

Enable game patching: (on)

Show NAND or user string in System Settings: (on)

Show GBA boot screen in patched AGB_FIRM: (on)

Patch ARM9 access: (off)

Set developer UNITINFO: (off)

Disable ARM11 exception handlers: (off)

--

**Explanation of the issue:**

Taking a screenshot in a VC game result in an incorrect color reproduction.
Color format detection works fine with 3DS games but it seems to be broken when it comes to VC.
See an example here: https://imgur.com/a/Ezeck.

**Steps to reproduce:**

1. Start a VC game.

2. Open the rosalina menu and take a screenshot.

**Dump file:**
-

jupitteer, 2017-09-16:
I can't seem to reproduce that, and I take a lot of screenshots.

Flaviculpus, 2017-09-16:
Added more examples. It looks like ntr doesn't have this issue. Also, why does it take so much time to take a screenshot using rosalina? Looking at how ntr does screenshots, it should be slower considering it works on a higher lever and uses nested loops, shoudn't it?

profi200, 2017-09-16:
Can reproduce this with the only GBC VC i have.
[image]

AuroraWright, 2017-09-24:
Fixed on https://github.com/AuroraWright/Luma3DS/commit/a39adc8ac13480077bb4d190e3325aefdb5f1204

closed 2017-09-24 by AuroraWright