LumaTeam/Luma3DS #855: Black Screen/ Don't boot home menu

scar86, 2017-09-16:
**System model:**

New 3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.3

**Luma3DS version:**
 8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Hello, yesterday I was playing as normal, my 3ds point me of a new version, but I just ignored the message, then I put the 3ds to sleep, today in the morning the 3ds don't boot into Home menu anymore, I can however boot the Luma3ds menu when I press start on boot, and other payloads as well.

I checked the FAQS and there is a section for black screen but luma 8.1.1 doesn't seem to have the "Enable exception handlers " anymore option to verify that was my issue

**Steps to reproduce:**

1. Turn on the 3ds

2.

**Dump file:**
No

Hikari-chin, 2017-09-16:
Cannot reproduce. Try follow Plailect's guide's [troubleshooting](https://3ds.guide/troubleshooting) section.

closed 2017-09-16 by Hikari-chin

HugaShire, 2017-09-16:
Restore nand backup. Highly unlikely that this is a luma issue. Sounds more of a Nand or SD issue. Also Exception Handlers are enabled automatically and can be turned off with "Disable ARM11 Exception handlers". Also check the 3ds.guide/troubleshooting.

scar86, 2017-09-16:
Thanks a lot you pointed me in the right direction, I followed part of the Plailect troubleshoot guide and after deleting the exdata, my 3ds worked again fine

RandyTheWolf, 2018-02-17:
Same here. Was on the way to follow Plailect's troubleshooting guide and it worked on it's own, showing the home menu.
For those who have a cheap sd card try to just wait it might work.

MarioMasta64, 2018-02-26:
sadly i wasnt able to boot through ntrboot, select, start, no sd, ninty safemode, or anything, no bootrom error just not booting after working perfectly the day before.