LumaTeam/Luma3DS #856: an exception error ocurred - data abort

Hayes231, 2017-09-16:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable): sysNANd: 9.2.0 emuNAND: 11.2**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS): boot9strap**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version: 8.1**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: (4 )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I was installing boot9strap on the sysNAND using soundhax. After saving the settings for Luma3ds it tried to boot but showed an exception error. I continued to the steps to move the emuNAND to the sysNAND. After finishing it still is not able to boot to either. 

**Steps to reproduce:**

1. power on 3ds

2.

**Dump file:**

first dump: 
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1308579/crash_dump_00000000.zip)

latest dump:
[crash_dump_00000013.zip](https://github.com/AuroraWright/Luma3DS/files/1308580/crash_dump_00000013.zip)

Hikari-chin, 2017-09-16:
Fairly common issue with emuNAND->sysNAND transfers I've noticed (though, I've never had it myself despite doing these often), and not really a Luma problem. Try [deleting the HM extdata or doing a ctrtransfer](https://3ds.guide/troubleshooting). If the issue persists, ask for help in the [NH Discord](https://discord.gg/C29hYvh).

closed 2017-09-16 by Hikari-chin

Hayes231, 2017-09-16:
Ok so I deleted the extdata like you said, and now i can boot to sysNAND! however, this is old sysNAND and not the emuNAND i was trying to transfer. .