LumaTeam/Luma3DS #862: Arm 11 An exception occurred

Piruy, 2017-09-20:
**System model:**
New 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39U
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1 stable
[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
When booting up Metroid: Samus Returns, a exception error screen pops up

Game error:
Processor: ARM11 (core 0)
Exception type: data abort
Current process: MATADORA (00040000001BB200)

**Steps to reproduce:**

1.Boot up Metroid

2.

Sonlen1414, 2017-09-20:
Seems to be a common issue with people who pirated the game. Buy it, it is worth the money.

hsuanpai, 2017-09-21:
Change the Region of Game to USA , Language of Game to EN. And Problem will be solved. Thanks.