LumaTeam/Luma3DS #867: Wierd bug in reigon emulation

trbocode, 2017-09-22:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S 1.2

**Luma3DS version:**

My own fork with different fonts- DEFENITLY NOT THE CAUSE.

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4?)

Splash: (none )

PIN lock: ( none)

New 3DS CPU: (Yes, L2 and MHZ )

--

Autoboot EmuNAND: ( X)

Use EmuNAND FIRM if booting with R: ( X)

Enable loading external FIRMs and modules: (X )

Enable game patching: (V )

Show NAND or user string in System Settings: (V )

Show GBA boot screen in patched AGB_FIRM: (V )

Patch ARM9 access: ( X)

Set developer UNITINFO: (X )

Disable ARM11 exception handlers: (X )

--

**Explanation of the issue:**

[image]
[image]

The first one is an American 3DS running an American MK7 copy. Reigon emulation is set to Jiangsu, China, and all the matching stuff.
The second one is an iQue running an American MK7 copy.
As you can see, both have bugs, but different kinds of bugs.

**Steps to reproduce:**

1.Set reigon emulation to Jiangsu (and the matching stuff)

2. Go to the MK7 channel

AuroraWright, 2017-09-23:
Region emulation doesn't change the game's code, so this is not Luma's fault.

closed 2017-09-23 by AuroraWright

trbocode, 2017-09-25:
But both games are the same...

ValliantStorme, 2017-09-25:
The iQue 3DS doesn't have the english language fonts. It's not a surprise things are broken; Luma doesn't try to fix font issues.

trbocode, 2017-09-26:
The font is clearly provided by the game.
The location on the american 3DS (First one) is wrong. That's the problem.
THINGS should look like the second picture, but they don't.