LumaTeam/Luma3DS #872: I've experienced multiple crashes over the past few months after installing apps.

Merik2013, 2017-09-24:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.6.0-39U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1.1-f6483ec

**Luma3DS configuration/options:**

Default EmuNAND: ( n)

Screen brightness: ( 4)

Splash: (y )

PIN lock: (n )

New 3DS CPU: ( always on)

--

Autoboot EmuNAND: (n )

Use EmuNAND FIRM if booting with R: ( n)

Enable loading external FIRMs and modules: (n )

Enable game patching: (y )

Show NAND or user string in System Settings: (y )

Show GBA boot screen in patched AGB_FIRM: ( y)

Patch ARM9 access: (n )

Set developer UNITINFO: ( n)

Disable ARM11 exception handlers: (n )

--

**Explanation of the issue:**

The system has been crashing occasionally in both arm9 and arm11. It started happening after I updated a number of my homebrew applications some time back, including my injected FBI. It seems to occur most commonly after installing new cias and then allowing the system to refresh. It may be FBI related, but since I've got Luma dumps from it I figured I should get your opinion on it and so we can determine the cause.

**Steps to reproduce:**

easiest method is this:
1. Install cia through FBI

2. open and close system settings so that the system refreshes itself. It will then crash.

**Dump file:**
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1327614/dumps.zip)

Hikari-chin, 2017-09-25:
Sounds like your system is bugged, certainly not a Luma issue. 

And, you shouldn't be using injected anything anymore, it's an obsolete dirty hack.

closed 2017-09-25 by Hikari-chin

jcmejia, 2017-10-20:
I got a question if I install cfw on my 3ds using ntrboothax would my
nintendo network id be lost or would it be intact after installation