LumaTeam/Luma3DS #873: Homebrew Crash Error

Sil3ntfury, 2017-09-24:
**System model:**

[New 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.1.1]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: (X)

Use EmuNAND FIRM if booting with R: (X)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
So when ever I try to load up homebrew this error keeps poping up.

[image]

**Steps to reproduce:**

1. ?

2. ?

**Dump file:**
(https://github.com/AuroraWright/Luma3DS/files/1327709/crash_dump_00000003.zip)

fincs, 2017-09-24:
Are you actually using Rosalina's 3dsx loading stuff? Keep in mind that hax 2.x is not supported.

Sil3ntfury, 2017-09-24:
Not sure. I did everything on the guide and then once it re-boots and I try to go to homebrew it does this.

fincs, 2017-09-24:
How are you actually launching a 3dsx?

Sil3ntfury, 2017-09-24:
Soundhax

Shadow322, 2017-09-24:
Use Rosalina to boot homebrew

fincs, 2017-09-25:
As I suspected.

closed 2017-09-25 by fincs