LumaTeam/Luma3DS #876: Error without touch any file

lordferky, 2017-09-26:
System model:

New 3DS (XL)

SysNAND version (+emu/redNAND version if applicable):

11.6.0-39 E SysNAND, 9.2-something EmuNAND (no longer actively in use)

Entrypoint (How/what you're using to boot Luma3DS):

Boot9Strap 1.3

Luma3DS version:

8.1.1 stable

Luma3DS configuration/options:

Default EmuNAND: ( 1 )

Screen brightness: ( 2 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Clock+L2 )

--

Autoboot EmuNAND: ( Off )

Use EmuNAND FIRM if booting with R: ( Off )

Enable loading external FIRMs and modules: ( Off )

Use custom path: ( I remain unsure that this feature still even exists in current builds )

Enable game patching: ( Off )

Show NAND or user string in System Settings: ( Of )

Show GBA boot screen in patched AGB_FIRM: ( Of )

Patch ARM9 access: ( Off )

Set developer UNITINFO: ( Off )

--

Explanation of the issue:

Always that i turn on on 3ds emnu start this error, i dont know how can i do, i cant see the menu more than 2 or 3 seconds

Dump file:
No dump file, I assume because it's an error, not a crash.

--
An error occurred (ErrDisp)

Error type:       generic

Process ID:       21
Process name:     qtm
Process title ID: 0x0004013000004202

Address:          0x00103c60 
Error code:      0xf96183fe

Sonlen1414, 2017-09-26:
QTM is Super Stable 3D, errors with that typically mean the hardware is dying or cables came loose, almost definitely not a Luma issue.

lordferky, 2017-09-26:
Joe can i fix It?

lordferky, 2017-09-26:
How can i fix It?

Sonlen1414, 2017-09-26:
Hardware issue, fine someone or somewhere that will check and replace/fix the 3D camera (I believe it was)

closed 2017-09-26 by Hikari-chin