LumaTeam/Luma3DS #880: Luma3DS v7.1 nonfunctioning payloads

Gooselex1, 2017-09-27:
N3DS XL, Black
11.5.0-38U SysNAND
sighax/boot9strap v1.0
Luma3DS v7.1 stable

Default EmuNAND: (X)
Screen brightness: (4)
Splash: (Before, also nonfunctional)
PIN lock: (Previously, currently disabled)
New 3DS CPU: (Off)
Autoboot EmuNAND: (NOT SHOWN)
Use EmuNAND FIRM if booting with R: (NOT SHOWN)
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: (They're currently NOT enabled, but that doesn't mean disabled...?)

**Explanation of the issue:**
Currently trying to update to b9s v1.2 to get luma v8.0. I've hit a brick wall (following 3ds.guide) on trying to get the SafeB9SInstaller.firm to register in the chainloader. I've even deleted GodMode9 and HourGlass9 (my two previous payloads) out of the luma\payloads folder. Am I missing a folder path, are they stored somewhere else?

If more info is needed please ask.

**Steps to reproduce:**
Unknown how to reproduce.

**Dump file:**
Not needed?

Committee, 2017-09-27:
Use the bin version os safeinstaller if you are on luma7

Margen67, 2017-09-27:
# THIS IS NOT A SUPPORT FORUM! For support go here:
# Nintendo Hacking: https://discord.gg/MjzatM8y

closed 2017-09-27 by Hikari-chin