LumaTeam/Luma3DS #881: Screenshots taken with Rosalina in the G/S (or injected Crystal) VC games have distortion

cetertel, 2017-09-28:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.3

**Luma3DS version:**

8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (2 )

Splash: (Before )

PIN lock: (Off )

New 3DS CPU: (Clock )
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Any screenshots I make in the G/S (or my inject of Crystal) end up looking like this: https://i.imgur.com/y8mnoSr.png

**Steps to reproduce:**

1. Play the VC games

2. Take a screenshot with Rosalina

3. Screenshot is corrupted

ValliantStorme, 2017-09-29:
Try the latest nightly release of Luma3DS

cetertel, 2017-09-29:
http://i.imgur.com/qSu0qvi.png

Looks like it's been fixed then!

closed 2017-09-29 by cetertel