LumaTeam/Luma3DS #882: new-hbmenu hangs on homebrew launching when injected in to H&S on a New 3DS

zaide-chris, 2017-10-01:
**System model:**
New 3DS XL
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39U
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
Luma3DS 8.1.1
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
new-hbmenu hangs on homebrew launching when injected in to H&S on a New 3DS, works fine on my OLD 3DS
new-hbmenu blames Rosalina fincs/new-hbmenu#23

**Steps to reproduce:**

1. Set Rosalina to inject Homebrew Launcher into H&S 

2. Try to start any .3dsx file

**Dump file:**

Hikari-chin, 2017-10-02:
Known issue, duplicate.

zaide-chris, 2017-10-02:
I have a question. If something is a known issue why doesn't it have an open issue?
So that people know not to report it again?

Alex131089, 2017-12-03:
@Hikari-chin I wanted to have more information and I couldn't find, which bug is it a duplicate of ?

urherenow, 2017-12-03:
This is because H&S doesn't have the permissions that new-hbmenu expects (such as wireless). How about you set it to download play, the the guide tells you to? And Rosalina doesn't inject anything, it only re-directs...

fincs, 2017-12-03:
Wrong. All service access checks, filesystem access checks (both ARM9 and ARM11) and ARM11 system call access checks are overturned by Rosalina and Luma3DS.

Alex131089, 2017-12-04:
And it is really about the injected hbmenu, not rosalina's hijack.
The bug seems to be occurring on O3DS: when loading an homebrew, the system hang (and eventually lead to a dump)

urherenow, 2017-12-04:
It will do the same on N3DS. It has only ever worked for me using download play.

fincs, 2017-12-04:
I think people are misunderstanding what the 3dsx loading mechanism does in Rosalina. It does **not** replace an official .app by an "injected" .app. It does **not** use the mechanism used by hax 2.x to load code, which is just to DMA homebrew code on top of an already loaded official app. It detects and hijacks the operating system's commands to build a codeset, create a process and do several permission checks. In short, when the "takeover" title is launched, Luma3DS's custom Loader implementation forwards the creation of the codeset to Rosalina, which then proceeds to build said codeset from the 3DSX file's own internal specifications. Additionally Rosalina provides Loader with a custom exheader that has access to everything; which is augmented by Luma3DS' patching of many access permission checks (also involving the recently introduced SM and PXI module reimplementations). This code can be [publicly](https://github.com/AuroraWright/Luma3DS/blob/master/sysmodules/rosalina/source/hbloader.c) [reviewed](https://github.com/AuroraWright/Luma3DS/blob/master/sysmodules/rosalina/source/3dsx.c).

Alex131089, 2017-12-04:
Oh, I reread the issue, and my bad. I read the issue a bit too fast & got mislead by the title, I though it was really about the injection (.app), not the title hijack (rosalina).
Sorry for the noise 😶
Except for the hbmenu loading method, the symptoms looks the same though (hbmenu starts then hang when loading homebrew/3dsx).

urherenow, 2017-12-04:
@fincs thanks for the information. I was pretty aware of the first part, but not about building a codeset using the 3dsx specs. I was almost going to ask if this was therefore a bug in the new-homebrew menu, but again, there are no issues with it setting Download Play as the target. H&S for this task has issues.

urherenow, 2017-12-12:
Now I'm confused. Just reread everything because the issue was reopened. Noticed that @fincs mentioned on the new-hbmenu issue that this was a Rosalina problem. Then comes here and makes it sound like Rosalina is NOT the problem? 🤔

fincs, 2017-12-12:
It *is* a Rosalina problem, hence why this issue was reopened.

TuxSH, 2017-12-26:
Tried it on Luma3DS 9.0, "injecting" new-hbmenu into H&S then launching FBI.3dsx when doing the setup for one of my family members this christmas.

Will close this issue if it becomes inactive.

NightYoshi370, 2018-02-02:
Uhh, this has been dead for a month. Shouldn't this be closed?

Hikari-chin, 2018-02-02:
No, this bug persists in even the latest commit, the issue will be closed once it's fixed.

NightYoshi370, 2018-02-02:
TuxSH said that this has been fixed though. Oh well.

Hikari-chin, 2018-02-02:
That was outdated information; the bug is only present on certain models as far as I'm aware. Our apologies for not clarifying earlier.

NightYoshi370, 2018-02-02:
Its alright.

Anyways, one thing we could try doing is introducing debugging logs to the new-hb application. However, I suppose that's outside the scope of Rosalina/Luma3DS, so I'll stop.

MrMonotone, 2018-08-21:
Is there anything that I can do to help debug this issue?

urherenow, 2018-08-21:
No. You can follow the guide and use download play, as it says. Or look at the countless clised issues with this very problem that were answered already.

3ds.hacks.guide

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH