LumaTeam/Luma3DS #883: Efecto congelado al actualizar un juego

tetradog, 2017-10-02:
[spanish]
Cuando actualizo un juego parece que se queda congelado y no me gusta...además si no hay espacio disponible para la actualización lo intenta igualmente...creo que no cuesta nada mirar cuanto ocupa y mirar si cabe...me gustaría que hubiera una forma de ver el estado actual de la actualización así puedo estar tranquilo.

Shadow322, 2017-10-02:
ENGLISH PLEASE

closed 2017-10-02 by Hikari-chin

tetradog, 2017-10-02:
When update some game, cant see any progress and feels like frozen the system,please can add any progress bar or something...

Other problem when update without MB available...dosent stop just try a until update...becouse never will can finish...

Sorry my bad English...