LumaTeam/Luma3DS #884: Update game, efect Froze

tetradog, 2017-10-02:
When update some game, cant see any progress and feels like frozen the system,please can add any progress bar or something...

Other problem when update without MB available...dosent stop just try a until update...becouse never will can finish...

Sorry my bad English...

Tyulis, 2017-10-02:
Where do you update games?

tetradog, 2017-10-02:
When i want to play, the system say update...(clic into icon game)

Hikari-chin, 2017-10-02:
This isn't a Luma problem, and this is infeasible to "add" to Nintendo's software.

closed 2017-10-02 by Hikari-chin

tetradog, 2017-10-02:
It's doesnt true because when i update this program call freeshop...i know this because I power off while updating and free ship say me this incident while updating...

AuroraWright, 2017-10-02:
We don't support piracy, and a freeze in a program is not Luma's fault (in this case it's most likely a GPU freeze if you don't see exceptions).

tetradog, 2017-10-02:
Not freeze literally is an effect when update some game...