LumaTeam/Luma3DS #885: Desactived Rosalina

kasai07, 2017-10-03:
`
#define DESACTIVE_ROSALINA_COMBO  (BUTTON_DOWN | BUTTON_X | BUTTON_SELECT | BUTTON_START)

void menuThreadMain(void)
{
    if(!isN3DS)
    {
        rosalinaMenu.nbItems--;
        for(u32 i = 3; i <= rosalinaMenu.nbItems; i++)
            rosalinaMenu.items[i] = rosalinaMenu.items[i+1];
    }
    else
        N3DSMenu_UpdateStatus();

    while(!terminationRequest)
    {
        if((HID_PAD & menuCombo) == menuCombo)
        {
            menuEnter();
            if(isN3DS) N3DSMenu_UpdateStatus();
            menuShow(&rosalinaMenu);
            menuLeave();
        }
		if((HID_PAD & DESACTIVE_ROSALINA_COMBO) == DESACTIVE_ROSALINA_COMBO)
        {
			return;
        }
        svcSleepThread(50 * 1000 * 1000LL);
    }
}
`

Derek1199, 2017-10-03:
What is this issue even about? Lol.

gnmmarechal, 2017-10-03:
wa?

Margen67, 2017-10-03:
@kasai07 At least close your issue.

closed 2017-10-03 by Hikari-chin

kasai07, 2017-10-04:
Lol sorry