LumaTeam/Luma3DS #886: arm11 core0 iron15 error

ishtiaq1234, 2017-10-06:
**System model:**

N3DS

11.6.0-39E sysnand

B9S

**Luma3DS version:**

8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: 1(x)

Screen brightness:4(x)

Splash: Before( x)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

used FEFEditor to edit some files in the games rom, put it in the correct directories its ment to be in to run but when i get to the first chapter the game always crashes at the exact point each time (see image below)

[image]

ValliantStorme, 2017-10-07:
It sounds like something went wrong with the files you've edited.

ishtiaq1234, 2017-10-07:
oh alright ill re edit my files see it that helps

piratesephiroth, 2017-10-07:
maybe you should try it in Citra first (if it can run that game)

Hikari-chin, 2017-10-10:
>used FEFEditor to edit some files in the games rom

Definitely not a Luma issue.

closed 2017-10-10 by Hikari-chin