LumaTeam/Luma3DS #887: LayeredFS Code.bin wrong acces.

Damin72, 2017-10-07:
**System model:**
Old 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39E SysNAND
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
B9S 1.3
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (0)

Screen brightness: (5)

Splash: (0)

PIN lock: (0)

New 3DS CPU: (0)

--

Autoboot EmuNAND: (0)

Use EmuNAND FIRM if booting with R: (0)

Enable loading external FIRMs and modules: (0)

Enable game patching: (1)

Show NAND or user string in System Settings: (0)

Show GBA boot screen in patched AGB_FIRM: (1)

Patch ARM9 access: (0)

Set developer UNITINFO: (0)

Disable ARM11 exception handlers: (0)

--

**Explanation of the issue:**
When i patch "Fire Emblem Echoes: Shadows of Valentia" with translated files - everything ok.
But when i try put "code.bin" from FEE:SoV Patch 1.1, from game disappear my edited font. 
Seems to be "code.bin" ignore font file from folder with translate and tool it from original game folder. 
Question: How to fix this? 

**Steps to reproduce:**

1.(Need EUR version of Fire Emblem Echoes: Shadow of Valentia with Update 1.1) Put folder from this archive (https://drive.google.com/open?id=0B9vr_WZsN50DVnUwZFZZS2lDOUk) in folder
00040000001B4100>romfs.

2. Look at the font. If it's normal (original) - that's a problem.

**Dump file:**

Tyulis, 2017-10-07:
I see 2 possibilities. You modified the wrong file (font is taken from another file, or path is incorrect), or there is a problem with the patched code.bin file. LayeredFS replaces original files by patched ones when romfs is mounted, so game cannot access to original ones.

Damin72, 2017-10-07:
If game can't access to original one, and font is correct (Because if delete code.bin from patch, it's work), then... Code.bin contain this font... Only logical explanation.

Tyulis, 2017-10-08:
Another: You edited the wrong file, try to check if there is not any similar file in the romfs (for different languages...). In any case, this is not a Luma issue, so this is not the right place to ask that.

Damin72, 2017-10-08:
My bad. This code.bin doesn't work without installed patch.

Tyulis, 2017-10-08:
Ok. So you can close.

closed 2017-10-08 by Damin72