LumaTeam/Luma3DS #888: Luma 3DS 8.1.1 crash at loading

fraf, 2017-10-07:
I explain the process I did before show you the crash that permanently occurs now.

I had an old version of luma 3ds (arm9 bin loader + menu hax).
I have sysnand 9.2 (downgrade)
I had emunand 11.5 (OK)

All was working and decided to install bootstrap9 now following YouTube tutorial : https://www.youtube.com/watch?v=Hgg2kuibs7g

So I formatted my SDCARD content *EXCEPT* "Nintendo 3DS" folder.
Then I took files from the link in the tutorial (which was updated because the options selector menu mentionned Luma3DS 8.1.1 instead of 7.1 in the video)
and extract all in SDCard exactly like in the tutorial.
I reached 3:30 minutes all was OK and when luma3ds seems to boot, a crash occurs.

A screen of the crash here  .. :

[URL=https://www.hostingpics.net/viewer.php?id=164260IMG20171007135804133.jpg][IMG]https://img11.hostingpics.net/thumbs/mini_164260IMG20171007135804133.jpg[/IMG][/URL]

At 3:10 minutes, safehax made a backup firm0firm1.bak in boot9strap folder. Do I have to restore Something ? Can I fix luma start ?
I'am blocked now.
I can only start bootstrap9 menu (holding start button) ... to do what with ?
I can erase luma config .. to do what with ?
Otherwise, I boot with the exception screenshot.

Starting without sdcard causes black screen then 3DS switch off automatically.

Thanks for your help (I don't know how to continue with this luma crash).

fraf, 2017-10-07:
I think my mistake is that I kept "Nintendo 3DS" folder from my old installation (ARM9 + menu hax).
I will attach a crash dump soon.

Tyulis, 2017-10-07:
Read the official guide on [3ds.guide](http://3ds.guide), there is a full, precise and up to date tutorial and troubleshooting

fraf, 2017-10-07:
here the dump :

https://nofile.io/f/3OG8jUqswpZ/crash_dump_00000000.dmp

I followed the part "Black screen on SysNAND boot after Installing boot9strap" because I have bootstrap install (holding start works and menu appears).
But Nothing works. I should go for CTRTransfert so ? But I have firmware 9.2, is it ok to restore NAND ? Links in the guide are for 11.5 doesn't it ?

Can I follow the guide ?

Thanks.

fraf, 2017-10-07:
Problem solved with troubleshooting guide as mentioned.
The problem was me !
I deleted the wrong folder into "Nintendo 3DS" folder. It was the one in emunand (32 number/letters name), not in sysnand (also 32 number/letters) I deleted.
My fault !

Thanks you very much, you can close !

closed 2017-10-07 by fraf