LumaTeam/Luma3DS #889: When booting Luma from CTRNAND, it ignores LayeredFS on SD Card

kftX, 2017-10-11:
**System model: O3DS**

**SysNAND version): 11.5**

**Entrypoint: B9S**

**Luma3DS version: Luma3DS-98d4345 Nightly (updated from 8.1.1 to see if issue was fixed)**

**Luma3DS configuration/options:**

Default EmuNAND: (No EmuNAND)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I spent the entire afternoon trying to understand why the LayeredFS folder to load the Pokémon Nova Sun romhack was not working, trying everything from reinstalling the game from the store down to updating Luma to the latest nightly.

Then a lightbulb went off in my head remembering that when loading from CTRNAND, Luma tends to ignore anything that is installed on the SD, from txt files to change system string to splash screens. So I assumed that this issue would also happen when trying to load a LayeredFS folder from SD:/luma/titles

While the splashes and system screen changing are minor issues, I feel not loading the LayeredFS folder from SD on CTRNAND can be a bit annoying since being free of a SD card to load Luma is much better than being tied to one.

**Steps to reproduce:**

1. Load boot.firm from CTRNAND;

2. Install a romhack into SD:/luma/titles;

3. Load romhacked game and notice you are playing the non-hacked one.

Margen67, 2017-10-11:
Not an issue. Luma loads from wherever it was booted from. (CTRNAND/SD card)

kftX, 2017-10-11:
Which makes no sense. If I prefer to have Luma on installed CTRNAND it shouldn't be searching for LayeredFS on CTRNAND but on SD, especially because CTRNAND has no space for romhacks/translations in LayeredFS anyway.

It's a small issue, but it's those small quality of life improvements that make software more painless to use. Having two installations of payloads/Luma is kind of a pain.

Shadow322, 2017-10-11:
As I was told and passing it on with the pathetic help here "put Luma3DS on sd card"

AuroraWright, 2017-10-12:
This is on purpose, there's no meaning in not having a config.bin in /luma if you have the romfs folders and everything.

kftX, 2017-10-12:
Hm, I get that you mean that this is by design. And while I can understand your rationale for doing it this way, it can be a bit of a bother. I spent an entire afternoon trying to understand why Nova Sun wasn't loading at all, only to realize that since Luma was located in CTRNAND it didn't bother scanning the SD for LayeredFS.

I'm not saying you should make this a priority at all, nor that you should immediately patch it. But for the sake of other users who may have this issues, I decided to leave this issue here as a suggestion/reminder should you ever want to do something about it.

As someone who loves using Luma, I just felt like I had to do my part after coming across a minor issue.

Hikari-chin, 2017-10-12:
I think what @AuroraWright is getting at here is, if you already have /luma on your SD card with config.bin, LayeredFS, etc. stuff on it, why is having boot.firm on your SD card an issue? The LayeredFS files themselves fall victim to the exact same justification you used for boot.firm on SD being an inconvenience. 

Additionally, Luma Update and MultiUpdater both are capable of updating payloads on the SD and CTRNAND, with Luma Update doing both automatically (if you enabled the feature; it asks you if you want to), so I don't see the issue of having two boot.firms in different locations.

closed 2017-10-12 by Hikari-chin

kftX, 2017-10-12:
I just think it's more busywork than needed when you could just have boot.firm on the CTRNAND and *never* needing it on a SD. I just had to make a second Luma install on the SD specifically to be able to use LayeredFS romhacks because it didn't want to detect them on the SD (since CTRNAND Luma only scans CTRNAND for Luma related files). I was pretty happy only having CTRNAND Luma, now I'm forced to have CTRNAND and SD just for one feature.

As I said it is a bit of a pain, but oh well. Thanks anyway :)

Hikari-chin, 2017-10-12:
I'm sorry, but I don't see dropping an additional file on the SD card as a problem, when you've already dropped hundreds-thousands for LayeredFS, that is surely more of a pain when swapping SD cards than transferring over a single small file is.