LumaTeam/Luma3DS #891: [Suggestion] Add external fdpver loading.

mariohackandglitch, 2017-10-12:
With commit 76f057d the friends module patch was removed as it was considered "useless". The reason given was that the latest fpdver was hardcoded and had to be changed each time Nintendo released an update.

The suggestion is: Instead of hardcoding the fpdver in the source, it could be stored in a txt file on the SD. Thanks to this, a homebrew app could be made to check and update this file automatically each time a firmware version is released. The latest version number could be stored in a file in a github repository, for example, so the dev of the app would only have to update that file instead of the whole app or Luma itself.

You may think the patch is useless, but maybe not for some people. (Those running dex in 9.2, those who can't update because their wifi doesn't work for some reason, those that want to research lower firmware versions, etc)

Margen67, 2017-10-13:
Or you could stop using Gateway so you're not stuck on an ancient firmware

mariohackandglitch, 2017-10-13:
I don't have gateway.

movr1r1, 2017-10-13:
idk if Gatewait users will ditch Gateway or luma with that, we'll see.

AuroraWright, 2017-10-13:
I don't really know about what "dex" is, could you explain?
Also, the removal of this patch only makes it impossible to play games online. Hence it was of no use to people with no wifi connectivity.
And you can still boot old fws for research and whatnot, you just can't play online on them.
We deemed it useless not only for that, but because if you keep the console connected it will download the update anyway and then nag you to update constantly; and because there are no advantages whatsoever to staying with an older fw nowadays (no difference in game or homebrew compatibility, cfw features are the same...)

mariohackandglitch, 2017-10-13:
"dex" is a (closed fork) of NTR CFW with some debugging features such as memory watchpoints. (See https://github.com/AuroraWright/Luma3DS/issues/697). The problem is that it was designed for 9.2 and never updated again.

While you can downgrade to 9.2, it would be a pain to do it every time you want to play online again for whatever reason. 

Anyway, if this issue only involves my case, then I can make a pull request and do it myself, but I'd like to get some sort of green light before doing it and not waste some time.

closed 2017-10-14 by Hikari-chin