LumaTeam/Luma3DS #892: 3ds Smash Cartridge error

ssboss, 2017-10-14:
Old 3DS

System Version: 11.6.0-39U SysNAND

Entrypoint: Boot9Strap

**Luma3DS version:**
Luma3DS version 8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off )

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
When I fry to run the mod through Smash Selector, It won't let me use cartridge,(I can only use cia) and when I use cia, it doesn't load the mod and loads the game without any mods installed.

**Steps to reproduce:**

1.Turn on 3ds and open Smash Selector

2.Launch game with settings kept from boot up except change cartridge to cia

**Dump file:**
The error message DIDN'T tell me where the dump is even though I have Disable ARM11 exception handlers checked.

cetertel, 2017-10-14:
I had a similar issue a little while ago, try using JKSM to backup your save data, and then delete it by holding A, B, X, and Y upon startup. Try running your mods after that. If they work after that, you can try restoring your backup and see if they still work, otherwise you'll just need to use a fresh save file to load mods.

If for whatever reason this doesn't work, just restore your backup and no harm done.

ssboss, 2017-10-14:
Ok thanks! Let me try it.

ssboss, 2017-10-14:
It didn't work. Smash Selector just loads up vanilla smash even after doing what you said.

Hikari-chin, 2017-10-14:
Sounds like an issue with Smash Selector to me.

closed 2017-10-14 by Hikari-chin

ssboss, 2017-10-14:
Oh ok.