LumaTeam/Luma3DS #893: An exception ocurred while loading decrypt9WIP

ronpetit, 2017-10-14:
[Ol
[crash_dump_00000004.zip](https://github.com/AuroraWright/Luma3DS/files/1384798/crash_dump_00000004.zip)

d 3DS XL]

11.6.0-39U SysNAND

Boot9strap 1.3

[Luma 8.1.1]

Luma3DS configuration/options:

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

Autoboot EmuNAND: (Yes)

Use EmuNAND FIRM if booting with R: (No)

Enable loading external FIRMs and modules: (No)

Enable game patching: (No)

Show NAND or user string in System Settings: (Yes)

Show GBA boot screen in patched AGB_FIRM: (No)

Patch ARM9 access: (No)

Set developer UNITINFO: (No)

Disable ARM11 exception handlers: (

No)

Explanation of the issue:

With the decrypt9WIP.3ds file on the folder 3ds/Decrypt9WIP/, I ran the homebrew launcher and then select the Decrypt9WIP option, then the app crashes.

Steps to reproduce:

1. Copy the Decrypt9WIP.3ds file on the folder 3ds/Decrypt9WIP/
2. Open the homebrew launcher app and start Decrypt9WIP

[image]

fincs, 2017-10-14:
1) Use Luma3DS's chainloader feature to run arm9 apps instead of relying on old exploits.
2) Decrypt9WIP is deprecated in favour of GodMode9.

closed 2017-10-14 by fincs

ronpetit, 2017-10-14:
Is not that simple, for some reason, Luma is not loading anything with the chainloader feature, it just not respond @fincs

fincs, 2017-10-14:
The chainloader feature should be working. What file are you trying to load? Remember that it must be a `.firm` file.

Sonlen1414, 2017-10-14:
You aren't using the chainloader in the error message, that shows using a .3dsx version which is built for homebrew launch use. Chainloader is holding Start when turning on the console, and as fincs pointed out, .firm, not .3dsx.