LumaTeam/Luma3DS #895: Luma3DS Commit 95d2d0a Compilation Error

bennyman123abc, 2017-10-14:
**System model:**

* New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

* 11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

* Sighax

**Luma3DS version:**

* 8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (*)

Show GBA boot screen in patched AGB_FIRM: (*)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Luma3DS Compilation Issue:**

I am having trouble compiling the commit, [95d2d0a](https://github.com/AuroraWright/Luma3DS/commit/1a39cb27e46f69c3c3542b781b71c65084e0ce7b) currently. The issue I am having is with the Makefile and makerom. I have tried the latest release of makerom from the included ctrtool repository in the README.MD and I've tried compiling from the source. Here is the compile error.

```
make[1]: Entering directory `/c/Users/benny/Documents/Luma3DS-master/sysmodules/loader'
[SETTING ERROR] Unrecognised argument '-nocodepadding'
[RESULT] Invalid arguments, see 'c:\ctrtool\makerom.exe -help'
make[1]: Leaving directory `/c/Users/benny/Documents/Luma3DS-master/sysmodules/loader'
make[1]: Entering directory `/c/Users/benny/Documents/Luma3DS-master/sysmodules/rosalina'
[SETTING ERROR] Unrecognised argument '-nocodepadding'
[RESULT] Invalid arguments, see 'c:\ctrtool\makerom.exe -help'
make[1]: Leaving directory `/c/Users/benny/Documents/Luma3DS-master/sysmodules/rosalina'
cat build/loader.cxi build/rosalina.cxi > build/modules.bin
cat: build/loader.cxi: No such file or directory
cat: build/rosalina.cxi: No such file or directory
make: *** [build/modules.bin] Error 1
```

My compilation environment as follows:
* Microsoft Windows 10 Version 1703 (OS Build 15063.674)
* devkitPro 1.6.0
* ctrtool 0.15
* firmtool Latest Commit

(Please let me know if you need any more information about my compilation environment)

**Steps to reproduce:**

1. **Download commit from link above.**

2. **Replicate development environment (Should be the same).**

3. **Run 'make' in the extracted commit folder.**

Hikari-chin, 2017-10-14:
Your makerom is out of date.

closed 2017-10-14 by Hikari-chin

bennyman123abc, 2017-10-14:
What are the compile requirements for makerom and ctrtool? I can't get it to compile. I get this error:

```
oschar.c: In function 'os_fstat':
oschar.c:12:17: error: storage size of 'st' isn't known
  struct _osstat st;
                 ^~
oschar.c:12:17: warning: unused variable 'st' [-Wunused-variable]
oschar.c: In function 'os_fsize':
oschar.c:18:17: error: storage size of 'st' isn't known
  struct _osstat st;
                 ^~
oschar.c:18:17: warning: unused variable 'st' [-Wunused-variable]
oschar.c: In function 'os_fstat':
oschar.c:14:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
oschar.c: In function 'os_fsize':
oschar.c:23:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make: *** [oschar.o] Error 1
```

Hikari-chin, 2017-10-14:
You'd have to ask Project_CTR devs about that, preferably not on the Luma3DS issue tracker as it isn't the right place for it, but anyway, the Luma3DS README links to precompiled versions of makerom for Windows/Linux/macOS.