LumaTeam/Luma3DS #897: "An exception occurred" error when using Smash mods

ssboss, 2017-10-14:
Old 3DS

[e.g. 2DS, New 3DS, Old 3DS]

11.6.0-39U SysNAND

Boot9Strap

**Luma3DS 8.1.1**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
After I turn my 3DS on, open Smash Selector, and turn on SaltySD, I launch the game with the set settings. However, after pressing A for launch game, I get an error saying "An exception occurred" and am forced to shut down. 

**Steps to reproduce:**

1. Turn on 3ds and open Smash Selector

2. Launch game and keep settings except for enabling SaltySD. (because it's disabled on startup)

**Dump file:**
[crash_dump_00000017.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1385065/crash_dump_00000017.dmp.zip)

Tyulis, 2017-10-15:
If the crash dump says "data abort", it's because the problem is in the edited files, not in Luma.

ssboss, 2017-10-15:
Ok thanks! I'm going to try and reinstall the mods.

ssboss, 2017-10-15:
It didn't work. The same response came up and it told me to shut down.

Tyulis, 2017-10-15:
Ask this to the creator of the mod, it's not the right place for this issue.

ssboss, 2017-10-15:
But it happens with any mod I use, that's why I made it an issue here.

ssboss, 2017-10-15:
Never mind. I figured out the problem.

Tyulis, 2017-10-17:
See at the SaltySD repo, there is plenty of similar issues. Definitely not a Luma issue.

Hikari-chin, 2017-10-18:
Duplicate, not a Luma issue.

closed 2017-10-18 by Hikari-chin