LumaTeam/Luma3DS #898: Feature Request: Allow removal of files via RomFS patches

mariomadproductions, 2017-10-15:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: (x)

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Feature Request: Allow "removal" of files via RomFS patches. Maybe by (re?)adding support for .romfs file or a feature that reads a text file with all the paths of files that the user wants "removed".

**Steps to reproduce:**

N/A

**Dump file:**

N/A

Hikari-chin, 2017-10-18:
Unless your game has an internal handler for missing files or you edit its code.bin to no longer reference said files, the game would likely just crash. For this reason, I do not feel this is appropriate for Luma. 

However, please let me know if I'm misunderstanding what you're saying, because even if this were possible I don't see why packing all the files into a single file called romfs a la HANS would make this work.

closed 2017-10-18 by Hikari-chin

mariomadproductions, 2017-10-19:
I want to remove files because I suspect the game does have a internal handler for missing files.

I assumed that the Luma/HANS replaced the game .romfs with the sd .romfs but I guess I assumed incorrectly.

mariomadproductions, 2017-12-14:
How is this invalid?