LumaTeam/Luma3DS #899: [Feature request] EmuNAND manager

thenets, 2017-10-16:
It's too complicate manage EmuNAND with Luma3DS + GodMode9.
I don't know if is better add an EmuNAND manager to Luma or create a new payload just for it.
I was thinking about creating a "SafeMode9" for features like manage EmuNAND that will not brick the console.

DMSalesman, 2017-10-16:
Are you certain that this issue belongs to Luma?

thenets, 2017-10-16:
@DMSalesman I don't know. I was thinking that EmuNAND must be compatible with Luma, so the manager must be related to Luma or GodMod9 project.
Sorry if I posted in the wrong place.

Hikari-chin, 2017-10-18:
This is out of scope for Luma, belongs as its own payload.

closed 2017-10-18 by Hikari-chin

thenets, 2017-10-18:
@Hikari-chin @DMSalesman can you give me a way to follow? An example payload project will help a lot. And is possible run/test/debug payload with Citra?

DMSalesman, 2017-10-19:
@thenets 3DS development is out of my scope, I'm afraid.