LumaTeam/Luma3DS #903: Errdisp NFC

Seba201x, 2017-10-19:
**System model:**
New 3DS
**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39U Sysnand

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap
**Luma3DS version:**
8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When i use any app its shows ERRDISP Generic NFC, the only app i can use is system config

**Steps to reproduce:**

1.

2.

**Dump file:**

ValliantStorme, 2017-10-19:
Could you attach a crash dump?

Seba201x, 2017-10-19:
No because the 3ds doesnt dump, it shows this
[image]

ValliantStorme, 2017-10-19:
Huh, that is an issue

Seba201x, 2017-10-19:
I formatted and now i can't complete the config
[image]

Tyulis, 2017-10-19:
When did this issue happen te first time? Randomly, after Luma installation? And do you have access to GodMode9?

Seba201x, 2017-10-19:
After Luma instalation and yes, i have access

jcmejia, 2017-10-20:
Ok thanks

Tyulis, 2017-10-20:
Ok, so check if the issue is still here if you restore your NAND backup. It's possible that something was wrong in your installation.

Hikari-chin, 2017-10-21:
Can't reproduce; this seems like hardware damage; qtm (head-tracking) and other hardware parts also cause crashing if they're damaged.

closed 2017-10-21 by Hikari-chin

Seba201x, 2017-10-23:
but i changed the nfc antenna and i keep getting that error.

Tyulis, 2017-10-25:
Do you tried to restore your NAND backup?

Seba201x, 2017-10-25:
Finally i gived up and i bought another motherboard