LumaTeam/Luma3DS #904: Stuck in sleep after sleeping w/InputRedir on

SirNapkin1334, 2017-10-21:
[e.g. 2DS, New 3DS, Old 3DS]
O3DSXL
**SysNAND version (+emu/redNAND version if applicable):**
SysNand 9.2.0U; EmuNAND 11.6U (only tested on EmuNand); (Both have browsers)
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
b9s 1.2 (I think, either that or 1.3)
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1 Stable
**Luma3DS configuration/options:**

Default EmuNAND: (1) [Only have one, but I wish I had more, does anyone know how without the windows tool?]

Screen brightness: (4) [Does this honestly matter?]

Splash: (OFF)

PIN lock: (OFF)

New 3DS CPU: ( )
I'm on o3DS
Autoboot EmuNAND: (y)

Use EmuNAND FIRM if booting with R: (y)

Enable loading external FIRMs and modules: (n)

Enable game patching: (y)

Show NAND or user string in System Settings: (y)

Show GBA boot screen in patched AGB_FIRM: (y)

Patch ARM9 access: (n)

Set developer UNITINFO: (n)

Disable ARM11 exception handlers: (n)

**Explanation of the issue:**

If Rosalina InputRedirection is enabled, if you close the lid, upon reopening, the system will not wake from sleep mode. The screen remains black, and the power LED remains pulsing, like in sleep mode.
My friend tested this on 2DSXL and got the same results.

**Steps to reproduce:**

1. Open Rosalina menu, and enable InputRedirection.

2. Exit Rosalina menu.

3. Close lid (toggle sleep mode slider on o2DS)

4. Open lid (toggle sleep mode slider on o2DS)

5. Your system is stuck in sleep mode until you force it to power off by holding the power button.

**Dump File**
No crash.

SirNapkin1334, 2017-10-29:
I'm sure that this bug can be reproduced by anyone, so please try! We need some more people affirming that this happens.

Craftyawesome, 2017-11-03:
Mine worked the first close, but the second time the 3ds froze. Sleep mode still turned on/off the screen.

y0shimari0, 2017-11-10:
Can confirm, happened on first try

SirNapkin1334, 2017-12-28:
Also happens on N2DSXL (have tested personally).

Qyriad, 2017-12-28:
Old 3DS with Luma version 9.0 with B9S; I can also confirm my 3DS freezes after sleeping.

nsuchy, 2018-03-06:
It's worth mentioning this issue also occurs with NTR CFW if you are using streaming or input redirection and close the console. I wonder if something lower level is causing this.

Adrians1206, 2018-12-19:
> O3DSXL
> **SysNAND version (+emu/redNAND version if applicable):**
> SysNand 9.2.0-29U; EmuNAND 11.6-39E (tested on both)
> Also tested on 11.6-39U N3DSXL (same config)
> 
> **Entrypoint (How/what you're using to boot Luma3DS):**
> B9S 1.2 (I think, either that or 1.3)
> 
> **Luma3DS version:**
> 8.1.1 Stable
> **Luma3DS configuration/options:**
> 
> Default EmuNAND: (1) [Only have one, but I wish I had more, does anyone know how without the windows tool?]
> 
> Screen brightness: (4) [Does this honestly matter?]
> 
> Splash: (AFTER)
> 
> PIN lock: (OFF)
> 
> New 3DS CPU: ( )
> I'm on o3DS
> Autoboot EmuNAND: (y)
> 
> Use EmuNAND FIRM if booting with R: (y)
> 
> Enable loading external FIRMs and modules: (n)
> 
> Enable game patching: (y)
> 
> Show NAND or user string in System Settings: (y)
> 
> Show GBA boot screen in patched AGB_FIRM: (y)
> 
> Patch ARM9 access: (n)
> 
> Set developer UNITINFO: (n)
> 
> Disable ARM11 exception handlers: (n)
> 
> **Explanation of the issue:**
> 
> If Rosalina InputRedirection is enabled, if you close the lid, upon reopening, the system will not wake from sleep mode. The screen remains black, and the power LED remains pulsing, like in sleep mode.
> Input Redirection only has to be on, connecting to an external controller is not needed. My friend tested this on N2DSXL and got the same results. I tested this on another friend's N3DSXL and got the same results.
> 
> **Steps to reproduce:**
> 
> 1. Open Rosalina menu, and enable InputRedirection.
> 2. Exit Rosalina menu.
> 3. Close lid (toggle sleep mode slider on o2DS)
> 4. Open lid (toggle sleep mode slider on o2DS)
> 5. Your system is stuck in sleep mode, or at least a black screen, until you force it to power off by holding the power button.
> 
> **Dump File**
> No crash.

Try Updating luma to 9.1 and see if it crashes then. Thanks.

https://github.com/AuroraWright/Luma3DS/releases/tag/v9.1

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH