LumaTeam/Luma3DS #907: Some out-of-region games displaying incorrect dummy banners on other regions

FrootyRamn, 2017-10-23:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap

**Luma3DS version:**

8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: 1

Screen brightness: 1

Splash: Off

PIN lock: Off

New 3DS CPU: Off

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Some out-of-region games display incorrect banners on the Home Menu, due to the fact that games use different banners for different languages. If a proper banner for the user's language isn't needed, game devs tend to do one of two things: Using the default language banner for the languages that the game isn't intended for, or using a dummy banner that looks ugly and in the most extreme cases can make the game unusable. The solution usually ends up being the latter, which is a problem.

Example:

[Puyo Puyo Chronicle on a proper-region JPN 3DS](https://imgur.com/a/J4QQ3) vs [Puyo Puyo Chronicle on a USA 3DS](https://imgur.com/a/GHBay)

My proposed solution to this issue is having Luma make banners adhere to the language set in the region emulation settings for that particular game, meaning users who set the language to the one intended for the game will see the proper banner.

**Steps to reproduce:**

1. Insert or install an offending game (i.e. Puyo Puyo Chronicle or Puyo Puyo Tetris) in(to) a CFW 3DS that doesn't match its region (so, for my example, use a non-JPN 3DS).

2. Scroll to the game on the Home Menu to view its banner.

3. Gasp in awe as the logo for the game is missing. (you don't actually have to follow this step to replicate the issue)

AuroraWright, 2017-10-30:
I don't think this can be fixed, this is the home menu parsing the banner and only loading the banner of the appropriate region

closed 2017-11-22 by Hikari-chin

piratesephiroth, 2017-11-23:
Yeah this is a problem of the banner data being incomplete. Multiple regions are referenced but only the japanese one has the complete set of files.