LumaTeam/Luma3DS #908: System crashes and NAND gets corrupted after going home after exiting sleep mode.

hacker1024, 2017-10-23:
**System model:** New 3ds XL

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):** 11.5

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 8.1.1 stable, but I tried the latest commit at the time too

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (before)

PIN lock: (no)

New 3DS CPU: (Clock + L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
After leaving the system in sleep mode for at least a few minutes (or powering on while the system is closed), going home freezes the system and after rebooting the screens are black, but the backlights are on (and infrared light in the dark). Closing and opening turns the backlights off, and then pressing a button triggers a crash. The message is generic but the process says GSP. There's no dump. The only way to get the system in a working state again is to restore a NAND backup.

**Steps to reproduce:**

1. Turn on the system

2. Open a game

2. Put in sleep mode

3. After a few minutes, open again and try to go home.

**ALTERNATIVELY**
1. Turn the system on while closed

2. After a few minutes, open it.

**Dump file:** None

hacker1024, 2017-10-23:
A quick test and it looks like turning off Clock + L2 testing fixes it. I only tried once, though, it may not be related.

Hikari-chin, 2017-10-24:
Can't reproduce; it sounds like your system is damaged, either software or hardware-wise; I would try creating a completely new NAND, rather than restoring a backup, and seeing what happens.

closed 2017-10-24 by Hikari-chin

hacker1024, 2017-10-24:
Sorry to bother you, but how can I "create a completely new NAND?"

Hikari-chin, 2017-10-24:
The first thing I would try is a System Format, and if that's ineffective I would do a ctrtransfers (how-to available on 3ds.guide).