LumaTeam/Luma3DS #909: Can't access on Rosalina Menu

Clement95, 2017-10-23:
**System model:**

[3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[10.3.0-28E]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax]

**Luma3DS version:**

[8.1.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: (Off)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I can't access on Rosalina Menu because my L button is broken, can I change the combo to access on it ?

**Steps to reproduce:**

1. Press L+Down+SELECT but it doesn't work because my L button is broken

2.

**Dump file:**

urherenow, 2017-10-23:
Github isn't a support forum. You should ask on gbatemp or discord. I'm not holding my 3ds in my hand, but I'm pretty sure you have to access Rosalina to change the combo, so your choices are have someone change it and send you their config file, or modify the source and build your own default.

Hikari-chin, 2017-10-24:
From the Discord server, this should help you:

"This is the Rosalina config.bin to use if you can't open the Rosalina menu due to broken buttons. It changes the Rosalina menu combo to the "A" button. It's recommended that you change the Rosalina combo to something more convenient by opening the Rosalina menu, going to "Misc. Settings", and "Change menu combo"." 

https://cdn.discordapp.com/attachments/196635695958196224/369124653618298880/config.bin

closed 2017-10-24 by Hikari-chin

Clement95, 2017-10-24:
Thank you very much !

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Clement95, 2017-10-24:
Thank you very much !

Clement95, 2017-10-25:
...

Hikari-chin, 2017-10-25:
Try this one. 

https://cdn.discordapp.com/attachments/287345711492628490/372887450101612555/config.bin

Clement95, 2017-10-26:
I tried, I didn't find what is the Menu Combo, I tried "A" but it doesn't work

Clement95, 2017-10-27:
It's okay now, thank you !

FlamingNova, 2018-07-02:
I'm having a similar issue but neither config posted in this thread works when I copy it into the luma folder. It launches the config screen instead upon boot.