LumaTeam/Luma3DS #911: ARM9 error on any attempt at reading Titles from SD Card

adrot, 2017-10-25:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

8.1.1-f6483ec

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( 1)

Splash: ( Off)

PIN lock: (Off )

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
-Used ntrboot on said console to install b9s.
-Disabled Parental Controls, then updated (console was on Ver 3.0.0-6E).
-Installed FBI and homebrew, everything is fine.
-3 days later, ARM9 error apppears every time on boot (Version 8.1.1 stable)
-Made CTRTransfer, installed actual Luma
-Error "data abort" pops up when apps are trying to access Titles (Accesing Data Management>3DS>Software, Booting FBI, except injected, selecting the Titles section in injected FBI, opening Checkpoint)
-Another type of ARM9 error appears, other than "data abort", (When closing System Settings by short-prssing the Power button after accesing a homebrew and closing it, closing injected FBI on a Power button short-press), called "prefetch abort (svcBreak)"

**Steps to reproduce:**

1. Installing said Luma version on 3.0

2. Update to latest FIRM 

3. Atempt to run anything related to Titles. (Ex. Checkpoint)

**Dump file:**

[crash_dump_00000034.zip](https://github.com/AuroraWright/Luma3DS/files/1415393/crash_dump_00000034.zip)

Hikari-chin, 2017-10-25:
Can't reproduce the main issues, but the injection issue is known and is a problem of injection itself being well, a dirty hack. You shouldn't use injection any more, it's obsolete.

closed 2017-10-25 by Hikari-chin

adrot, 2017-10-26:
Issue has been solved personally. Corrupted Title data was found in the SD card. Thank you for your time, and I apologize for the wasted issue. You may now close this.

Hikari-chin, 2017-10-26:
Issue has already been closed, but enable the ARM9 exheader patches in Luma's config, that'll resolve FBI not working properly as a 3DSX in some scenarios.