LumaTeam/Luma3DS #914: Game patching: Replacing exheader.bin and romfs files causes an exeption

RicBent, 2017-10-29:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.0.0-33E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: 1

Screen brightness: 1

Splash: Off

PIN lock: Off

New 3DS CPU: Off
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I am replacing the exheaer.bin, code.bin and some files in the rom of New Super Mario Bros. 2 Gold Edition (0004000000137e00). The code.bin and exheader.bin files are the original files from the game. 
Without the exheader.bin file the exeption does not happen. It also does not happen if I try it with the code.bin and exheader.bin files only.

I have not tested this with other applications.

**Steps to reproduce:**

1. Replace a file from the romfs and the exheader.bin for an applicaion via the game patcher.

2. Launch the application.

**Dump file:**

[crash_dump_00000037.zip](https://github.com/AuroraWright/Luma3DS/files/1424728/crash_dump_00000037.zip)

Shadow322, 2017-10-29:
Update to latest 3DS fw version and get latest Luma3DS nightly

RicBent, 2017-10-29:
Updated to 11.6.0-39E. Still crashes.

The crash message is a bit different iirc so I uploaded another dump:
[crash_dump_00000038.zip](https://github.com/AuroraWright/Luma3DS/files/1425062/crash_dump_00000038.zip)

JorgeProGamer, 2017-11-02:
Try in the luma3DS config, seleted arm9 and arm11 options, sorry for my english, i am spanish.

JorgeProGamer, 2017-11-02:
Disable ARM11 exception handlers and Patch ARM9 access, select this options.

NightYoshi370, 2018-03-27:
Quick question:

Does this happen on a custom built CIA? And does this happen on other versions of NSMB2 as well?

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH