LumaTeam/Luma3DS #915: Can Luma3DS be provided in a .zip instead of a .7z ?

jason0597, 2017-10-31:
Due to the size of the files, compression isn't really needed. Providing it in a .zip would simplify it for a lot of people downloading Luma3DS as they wouldn't need a 3rd party tool to extract it.

Chriscox462, 2017-10-31:
You could always build from the .zip source code instead. The portable Version of 7zip is only 8 MB(3 MB installer).

urherenow, 2017-10-31:
You should'n get into the habit of building from .zip source code. You should always clone the repo using git. Most 3ds projects require --recursive cloning, and downloading a .zip of the repo simply won't get you there...

fincs, 2017-10-31:
> Most 3ds projects require --recursive cloning

What

urherenow, 2017-10-31:
like ntrboot_flasher, for instance (and numerous other projects)

If you don't "git clone --recursive https://github.com/....." you won't be able to build anything because you will be missing all of the required submodules. I understand it isn't required for Luma, I'm just saying that simply downloading a source  .zip from github is a bad habit to get into.

Hikari-chin, 2017-10-31:
This has been requested before and may change in the future, but currently there are no plans to do so as far as I'm aware.

closed 2017-10-31 by Hikari-chin