LumaTeam/Luma3DS #918: Instant Arm11 crash

Shadow322, 2017-11-03:
System model:

New 3DS

SysNAND version (+emu/redNAND version if applicable):

11.6.0U SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

boot9strap

Luma3DS version:

8.1.1 22a8661fe1840057d3a6a720b79659a85b50d633

Luma3DS configuration/options:

Default EmuNAND: 1

Screen brightness: 4

Splash: Off

PIN lock: Off

New 3DS CPU: Off

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Explanation of the issue:

Instant Arm11 crash

[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1439803/arm11.zip)

Hikari-chin, 2017-11-03:
Compiled this myself, cannot reproduce. Exact same hardware (New 2DS XL), version, config settings, etc.

Shadow322, 2017-11-03:
Then can u upload your Luma3DS .firm file plez

Hikari-chin, 2017-11-03:
https://a.uguu.se/so3Cq1mImXVw_boot.firm

TuxSH, 2017-11-03:
@Shadow322 by any chance, did you build it with an outdated dkA (run `arm-none-eabi-gcc -v` & paste the output to check)

Jerry-Shaw, 2017-11-04:
same here but under windows

fincs, 2017-11-04:
Outdated devkitARM. Please run the devkitPro Updater to get the latest version of devkitARM (r47).

Jerry-Shaw, 2017-11-04:
Wow, never update devkitPro before, thanks.

fincs, 2017-11-04:
devkit*Pro* is the organization that *pro*vides the devkits. You can't really install it. What you actually do is install devkitPro products such as devkitARM.

Shadow322, 2017-11-04:
@Jerry-Shaw can u contact me on gbatemp nick Shadow#1 I need some help since your on windows compiling.

Thanks

Jerry-Shaw, 2017-11-04:
@Shadow322 Yeah, posted to you, see ya on GBATemp.

Repflez, 2017-11-04:
I managed to replicate this consistently by using Boot NTR Selector 2.9 installed as CIA.

1. Open Boot NTR Selector 2.9.
2. Select any version or let it select 3.6 automatically.
3. Instead of returning to the HOME menu, it crashes.

On commit 8258a98, it gives an ARM11 crash, making a dump, of course, while on commit 22a8661 or newer it instead shows an ErrDisp message with the `0xd92103fb` error code. Commit 76dde0e is the last working commit.

Here's the dump given by by these steps:
[crash_dump.zip](https://github.com/AuroraWright/Luma3DS/files/1442955/crash_dump.zip)

Jerry-Shaw, 2017-11-04:
Yep, updating devkitPro works. Thanks guys.

closed 2017-11-04 by Hikari-chin