LumaTeam/Luma3DS #919: Luma3DS NAND issue

SSQ123, 2017-11-03:
**System model:**
2DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
10-5-0-30U SysNAND
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1 stable
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (before )

PIN lock: ( N/A)

New 3DS CPU: (Old 2DS )

--

Autoboot EmuNAND: (Y )

Use EmuNAND FIRM if booting with R: (N )

Enable loading external FIRMs and modules: (N )

Enable game patching: (N )

Show NAND or user string in System Settings: (Y )

Show GBA boot screen in patched AGB_FIRM: (N )

Patch ARM9 access: (N )

Set developer UNITINFO: (N )

Disable ARM11 exception handlers: (N )

--

**Explanation of the issue:**
Luma3DS doesn't show EmuNAND in my settings when I choose it to auto boot. IDK what I'm doing wrong.

**Steps to reproduce:**

1.

2.

**Dump file:**

Derek1199, 2017-11-03:
Either you don't have a emunand, or you fucked up your emunand.

Hikari-chin, 2017-11-03:
That, or you're using a custom version string .txt? Either way, can't reproduce; all 4 of my emuNANDs display the correct edited version text in Settings.

closed 2017-11-03 by Hikari-chin