LumaTeam/Luma3DS #920: Pokemon Super Mystery Dungeon arm11 Crash on Game Load Up?

DarKEmX, 2017-11-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

I'm guessing this is the "Payload type" or something. SIGHAX

**Luma3DS version:**

8.1.1-f6493ec

**Luma3DS configuration/options:**

Default EmuNAND: 1(x)

Screen brightness: 4(x)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Off(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Okay so every time I try to launch my physical cartridge copy of Pokemon Super Mystery Dungeon, the game launches and shows the three loading gears at the bottom right that are red, yellow, and blue. I assume the game finishes loading because the screen then fades to black but then this crash happens every single time. I can replicate the issue as many times as needed because, well, it doesn't go away and happens every time. What's going on? I'm not 100% fluent in the world of homebrew as I had a friend homebrew my n3DS for me while I watched and learned a few things so don't crap on me for not knowing certain things.

**Steps to reproduce:**

1. Launch Pokemon Super Mystery Dungeon.

2. Loading screen shows, and then crashes without even getting to the title screen first.

**Dump file:**

[crash_dump_00000026.zip](https://github.com/AuroraWright/Luma3DS/files/1444443/crash_dump_00000026.zip)

Hikari-chin, 2017-11-05:
This is typically due to corrupt save or update data. Either way, can't reproduce.

closed 2017-11-05 by Hikari-chin

DarKEmX, 2017-11-05:
I don't have any save data for the game. I've checked and I can't find any. If you could tell me where to locate any potential data that I can delete to fix it, that'd be nice.

Hikari-chin, 2017-11-05:
The issue tracker isn't for tech support; try the [Nintendo Homebrew Discord](https://discord.gg/C29hYvh).