LumaTeam/Luma3DS #922: New3DS Options show in Rosalina menu on Old2DS.

Lykie, 2017-11-06:
**System model: 2DS**

**SysNAND version: 11.6**

**Entrypoint: B9S**

**Luma3DS version:8.1.1-a4899a1b https://github.com/AuroraWright/Luma3DS/commit/a4899a1bec29a568552375074bbe839d4f4625a8**

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (X)

Enable game patching: (X)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

New 3DS menu option shows under the Rosalina menu on a O2DS.

**Steps to reproduce:**

1. Compile Luma, put it on the SD Card as boot.firm.

2. Turn on device and open Rosalina Menu.

Side Note: I'm actually currently unable to compile builds for myself, I grabbed the release from here https://astronautlevel2.github.io/Luma3DS/

Shadow322, 2017-11-06:
His builds r currently broken

Shadow322, 2017-11-06:
I have compiled my own build and as u stated the New 3DS features show up on O3DS

Hikari-chin, 2017-11-11:
Should be fixed in 44b5e10.

closed 2017-11-11 by Hikari-chin