LumaTeam/Luma3DS #925: Possible Regression

NWPlayer123, 2017-11-09:
I've been compiling all my own versions, I was using a version from October 16th (aka https://github.com/AuroraWright/Luma3DS/commit/95d2d0a6bd3b762bc6103c48b7d24ddde0e7edbd), just built and switched to latest commit (aka https://github.com/AuroraWright/Luma3DS/commit/a4899a1bec29a568552375074bbe839d4f4625a8).
I've been using the L+down+select menu to take screenshots, but since switching, 3 times it's just softlocked, only showed black screen and not the menu, had to hold power button to shut it off and turn it back on (I've opened the menu 30 times which means it isn't happening that often but it's still irritating).
I'd get a crash log if it actually made one, but since it freezes on opening the menu, I'm not sure what there is to do there.

bearprincesskai, 2017-11-10:
I am having the same issue.

olefb, 2017-11-11:
Does this version trigger the same issue for you? Built with all the latest libraries and toolchains. Does it happen anywhere you open the menu (home menu, in-game)? Which 3DS model do you have, and which Luma3DS options? No such issue here, after opening Rosalina on the home menu 50+ times. New 3DS XL, and the only Luma3DS options I'm running are Clock+L2 and game patching.
[boot-44b5e10.zip](https://github.com/AuroraWright/Luma3DS/files/1463655/boot-44b5e10.zip)

Hikari-chin, 2017-11-22:
I can't seem to reproduce this, took hundreds of screenshots of Stella Glow and Ultra Moon to no I'll effect.

closed 2017-11-22 by Hikari-chin