LumaTeam/Luma3DS #926: "An exception occurred" when loading Shin Megami Tensei IV

vgastaldi, 2017-11-11:
[Old 3DS]
11.6.0-39U SysNAND (CTRNAND)

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.3

**Luma3DS version:**
v8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: (X )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--
Whenever I click on the "Load game" option on Shin Megami Tensei IV an exception occurs. I'm using the eShop version of the game (downloaded directly from the canadian store). It is updated and only has one free DLC.

When I delete the save I am able to click on the load option without crashing. I tried to restore a save backup done in 25/01/2015, but the error returns.

**Steps to reproduce:**

1. Open Shin Megami Tensei IV

2. Select Load Game option

3. 3DS crashes

**Dump file:**
[crash_dump_00000010.zip](https://github.com/AuroraWright/Luma3DS/files/1463729/crash_dump_00000010.zip)

[Shin Megami Tensei IV save]
(https://github.com/AuroraWright/Luma3DS/files/1463730/sdds4game1.zip)

Hikari-chin, 2017-11-12:
Sounds and looks like your save is corrupted.

closed 2017-11-12 by Hikari-chin

vgastaldi, 2017-11-12:
Weird is that even the backup is corrupted...

But thanks anyway.