LumaTeam/Luma3DS #928: Feature Request: Allow removal of files via RomFS patches (2)

mariomadproductions, 2017-11-12:
See here: https://github.com/AuroraWright/Luma3DS/issues/898
There are games that have code to handle removed files.

Midnoclose, 2017-11-12:
Thi sis a good idea!

Hikari-chin, 2017-11-12:
If the game has missing file handlers, then this can already be accomplished by simply removing the files from the game's romfs folder on the SD card. LayeredFS works by mounting the romfs on the SD card as the game's romfs instead of the built in one, so the text file solution really isn't feasible; the only way to do this is the way I described.

closed 2017-11-12 by Hikari-chin

mariomadproductions, 2017-11-14:
@Hikari-chin 
That can't be true. If what you are saying is true, then if I put a single music file in the appropriate folder to replace a piece of a game's music, then the game would crash, because it would only have one file in its RomFS. If this was true, then that would mean every mod would require a copy of the whole game.

Hikari-chin, 2017-11-14:
You're right. I was thinking of an older, inferior implementation.

mariomadproductions, 2017-11-15:
Even an option to use that inferior implementation would be useful.