LumaTeam/Luma3DS #929: Smash crashes if you have Clock+L2 enabled

CCF100, 2017-11-13:
**System model:**
New 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39U SysNAND
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9strap 1.2
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v8.1.1
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: (x )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Game crashes if you have Clock+L2 enabled in Luma settings
[crash_dump_00000088.zip](https://github.com/AuroraWright/Luma3DS/files/1467319/crash_dump_00000088.zip)

**Steps to reproduce:**

1. Enable Clock+L2 in Luma config

2. Game crashes on launch

**Dump file:**

Chriscox462, 2017-11-13:
Does it crash if you turn it off?

Hikari-chin, 2017-11-13:
Some games have problems with forcing a certain clock mode; this isn't a Luma issue and isn't fixable.

closed 2017-11-13 by Hikari-chin