LumaTeam/Luma3DS #930: Newly installed Luma 8.1

cyberexodus, 2017-11-14:
So I have a O3DS with 9.5.0.22U update and I was follwing this tutorial on to installing cfw to my 3ds but I keep getting this screen after running soundhax.
This is the tutorial I used

https://3ds.guide/installing-boot9strap-(homebrew-launcher)

Shadow322, 2017-11-14:
what screen?

cyberexodus, 2017-11-14:
Sorry I'm new I was using mobile earlier. There's an update to my original.

Shadow322, 2017-11-14:
Did u update to 11.6?

cyberexodus, 2017-11-14:
No

Shadow322, 2017-11-14:
B9s installed?

cyberexodus, 2017-11-14:
Yes

Shadow322, 2017-11-14:
Then just update the 3ds

cyberexodus, 2017-11-14:
Can't start up soundhax,but luma is still active.

Shadow322, 2017-11-14:
U don't need soundhax anymore

cyberexodus, 2017-11-14:
Okay, but lets say I want to install cias what do I do?

closed 2017-11-14 by cyberexodus

urherenow, 2017-11-14:
use the rosalina menu to launch the homebrew browser. Run FBI.3dsx from that. You can install the .cia version of FBI onto your home menu from there. Also the hblauncher_loader if it meets your fancy.

cyberexodus, 2017-11-14:
From the rosalina menu what command do I set to enter the homebrew browser.

cyberexodus, 2017-11-14:
Lol just got it thanks for all the help guys 😆

Derek1199, 2017-11-14:
https://3ds.guide ...follow this.