LumaTeam/Luma3DS #932: Won't boot with boot.firm

TheRasPiGuy, 2017-11-16:
Ok, so I followed [this](https://3ds.guide/a9lh-to-b9s) tutorial to upgrade from A9LH to B9S, but after I finished everything, it worked for a while, and then suddenly, it wouldn't turn on. I took out the SD card to see if it would work, and it booted, so then I put my SD card back in, and it still worked, but when I tried to boot with the SD card back in, the blue power led turned on, and then a second later, it turned off. So I tried experimenting on my computer to see what file(s) were causing this, but when I moved the boot.firm onto my computer it booted normally... So, my question is, why does my O3DS on 11.6.0-39U hate the boot.firm file?
Also, I tried using the released boot.firm, and also tried modifying the source code and building it, but it still wouldn't boot.

Derek1199, 2017-11-16:
Install b9s 1.3, simple as that. And use the latest luma release.

TheRasPiGuy, 2017-11-16:
The thing is, I did, but I will try updating...

TheRasPiGuy, 2017-11-16:
I can't even update it because I can't get to the chainloader menu. It goes straight to GodMode9.
i hate my 3ds...

TheRasPiGuy, 2017-11-16:
wow, now it boots with sd card...

closed 2017-11-16 by TheRasPiGuy