LumaTeam/Luma3DS #934: n3DS crashes all the time no matter what settings in Luma are

A-Fat-Duck, 2017-11-17:
New 3DS XL (Eur/Aus)

**SysNAND version (+emu/redNAND version if applicable):** 11.6.0-39E 

**Entrypoint (How/what you're using to boot Luma3DS):** B9S

**Luma3DS version:** 8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

Autoboot EmuNAND: (n)

Use EmuNAND FIRM if booting with R: (n)

Enable loading external FIRMs and modules: (n)

Enable game patching: (tried on and off)

Show NAND or user string in System Settings: (n)

Show GBA boot screen in patched AGB_FIRM: (n)

Patch ARM9 access: (n)

Set developer UNITINFO: (n)

Disable ARM11 exception handlers: (tried on and off )

--

**Explanation of the issue:**

Copy of what I said on /r/3dshacks:

<3dshacks>
Eur/Aus new 3DS unit, I noticed this crashing soon after updating from A9LH to B9S and couldn't find anything similar so I gave up and hadn't touched the unit in months thinking an update would help, but even after Luma 8.1.1 recently the problem persists (and it seems to crash more & in different places with patching enabled in Luma). Fire Emblem Fates and Pokemon Sun are both at the stage where I cant progress any further without them locking up and being forced to restart, on top of other games crashing at random. Any ideas what it could be or how to solve this?

I've tried using another SD by copy and pasting my old SD card, but after that the games wouldn't load at all so I went back to the old SD and updated to 11.6 but that also didn't seem to make any difference.
https://imgur.com/a/ajQkj
</3dshacks >

on r/3dshacks it was suggested to try to disable the ARM11 exception handlers to see if the system would carry on with a minor error but the error codes still occurred (https://i.imgur.com/03MFrNd.jpg) and was told "Looking at your screenshots, it seems like the before and after exceptions look the same. You shouldn't see them if they've been disabled. :/" and to ask here what was happening

fincs, 2017-11-17:
```
<fincs> !err 0xd900458b
<plutobot> [3DS] Module: FS (17) | Description: RomFS hash-check failed? (fs) (395) | Summary: Wrong argument (8) | Level: Permanent (27)
```

Looks like your files on SD card are corrupted.

Sonlen1414, 2017-11-17:
The error code is a File System error, which would also explain why disabling the exception error handler didn't work, as it isn't booting off SD due to errors, backup what you can, format the SD, and copy stuff back. Those games may require repairing in eShop or reinstalling to fix the errors.

A-Fat-Duck, 2017-11-17:
@fincs What's the best way to backup the saves to easily restore them again? And if I get a new SD card/format the current one what folders/files do I need to put on again?

@Sonlen1414 I tried to copy and paste from one SD to another which seemed successful, but after that the games wouldn't even boot at all on the other SD (never had any problems with that one in my camera so I think it's ok) so would it be worth trying to format the original one and copy everything back onto it as is?

Shadow322, 2017-11-17:
If your just swapping sd cards no need to backup saves just copy everything to new card

urherenow, 2017-11-17:
What is the file format of the new card? Fat32 or exFat? What's the cluster size?

A-Fat-Duck, 2017-11-17:
@urherenow Fat32 (didn't think exFat worked on the 3DS?) and not sure on cluster size as I just used the default. Is there any particular cluster size that works best (I thought that this didn't matter all that much)?

@Shadow322 copying everything off the current SD card now so will report back after formatting it and putting all the data back (might try format and copy onto my spare SD again to test first in case)

urherenow, 2017-11-17:
No, exfat doesn't work, but the question needs to be asked. Some people either don't know that, or didn't think to check. Cluster size *shouldn't matter but it seems in some cases 32k works better than 64k with homebrew.

AuroraWright, 2017-11-17:
If the files are corrupted copying will do no good, you need to redownload the games.

A-Fat-Duck, 2017-11-17:
@AuroraWright I've backed up all the saves and tried searching around for a guide as to what to do for a fresh SD card but all I've found is talking about copying the old data onto a new one for replacing a smaller sized one. What will I need to put onto the formatted SD card for CFW before using the fresh SD?

closed 2017-11-17 by Hikari-chin