LumaTeam/Luma3DS #935: Console Freezes when Luma Input Redirection fails to start or looses connection.

SmackBro, 2017-11-17:
In addition to the "Stuck in sleep after sleeping w/InputRedir on #904"

N3DSXL
SysNAND version (+emu/redNAND version if applicable):
SysNAND 11.6.0.39

Entrypoint (How/what you're using to boot Luma3DS):
B9S 1.3

Luma3DS version:
8.1.1 Stable
Luma3DS configuration/options:

Default EmuNAND: (1) [Don't use emunand]

Screen brightness: (3) 

Splash: (BEFORE)

PIN lock: (OFF)

New 3DS CPU: (Clock +L2,but happened on "Off" too)

Autoboot EmuNAND: (n)

Use EmuNAND FIRM if booting with R: (n)

Enable loading external FIRMs and modules: (n)

Enable game patching: (y)

Show NAND or user string in System Settings: (y)

Show GBA boot screen in patched AGB_FIRM: (n)

Patch ARM9 access: (n)

Set developer UNITINFO: (n)

Disable ARM11 exception handlers: (n)

Explanation of the issue:

The System freezes if Input Redirection fails to connect [When I access the rosalina menu too fast after getting into the home menu or trying to start it when I'm playing Pokemon Ultra Moon],I still can get into the rosalina menu to reboot the console afterwards it is working like normal.
This also happens if I somehow loose the connection Ingame.

Steps to reproduce:

1. Boot into the home menu
2. Input the key kombo to bring up the rosalina menu before the upper bar of the touchscreen shows up
3. Start the debugger and wait until it fails to start
4. Close the menu [The system should work normally] bring up the rosalina menu again and close it [the system now should freeze]
6. Additional: This also happens if you flic the wifi switch/set the option it to "no" in the home menu settings on n3ds

As I said this also happens if you play while Input Redirection is active and it looses connection when you are ingame.

Hikari-chin, 2017-11-17:
Confirmed.

Nomelas, 2018-08-14:
happens to me as well even on latest build (9.1)

TuxSH, 2020-04-15:
Known issue.

Closing, tracker cleanup

closed 2020-04-15 by TuxSH