LumaTeam/Luma3DS #936: An Exception Occursed: Pokemon Moon Patching

BizarreDio, 2017-11-18:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.6.0-39U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.1.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (Before)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Dumped .cia file of Pokemon Moon, extracted romfs and exefs of it, modified it with pk3DS and placed folders inside of /titles/[Title ID] folder. When loading Pokemon Moon, I get [this error message.](https://i.imgur.com/k8qEE9k.jpg)

**Dump file:**

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/1484335/crash_dump_00000001.zip)

Tyulis, 2017-11-18:
The structure of the luma/titles/[TIDlow] should be:
romfs: edited files in the romfs
code.bin
(needed extheader.bin and locale.txt)

You cannot replace the entire exefs (i think)
If the error persist, it's a problem in your edited files. Not a luma issue.

BizarreDio, 2017-11-18:
Well the program I use to extract the CIA files bugs out on the newest version and the only version that successfully extracts romfs and exefs, Luma crashes on reading it. Using other romhacks works fine so what program can I use to extract .cia files? I use PackHack and it doesn't work.

Tyulis, 2017-11-18:
Extract the CIA directly using GodMode9

BizarreDio, 2017-11-19:
I extracted the CIA using GodMode9 and used PackHack to extract the content, both before and after encrypting the CIA. Luma still crashes on reading it, and I'm following every step precisely.

BizarreDio, 2017-11-19:
[This is the setup for my folder in the titles folder](https://i.gyazo.com/6c2fc1de2214ad0a3b5c495e5891a733.png).

Tyulis, 2017-11-19:
I don't think it has an influence, but remove banner.bin and icon.bin. Then, the .code file should be named code.bin and not .code.bin. And finally, why using PackHack?
In Godmode9, select the CIA, select Mount image to drive and then, you should see a folder named like "00000000", copy its content, it's the game files. If there is no folder like that, it's because the CIA itself has a problem.
Another possiblility: From where have you dumped the CIA? If it's from the cartridge or the .app file in SysnandSD/titles/<titleID>/00000000.app, it's also possible that you have got a conflict with the update. If it's the case, dump the content of the .app or .3ds itself, and then go to Sysnand SD/titles/0004000e/<titleid> and dump also the files contained in 0000000x.app, overwriting previous files.

BizarreDio, 2017-11-19:
I mounted the image and found the folder with the game data and I'm moving it to the SD Card now. I'll see if Luma works with this extracted romfs and exefs.

BizarreDio, 2017-11-20:
The mounted image produced a romfs, exefs, exefs.bin, extheader.bin, logo.bin, ncch.bin, plain.bin and romfs.bin, 

With the .code file, it wasn't a bin file this time and I haven't tried booting it up into Luma yet.

Hikari-chin, 2017-11-20:
Duplicate; not a Luma issue. PK3DS has known problems, though this particular one seems to be user error.

closed 2017-11-20 by Hikari-chin

Tyulis, 2017-11-20:
Rename .code into code.bin, you can then copy it in luma/titles/tid

BizarreDio, 2017-11-23:
Same issue is still happening, using both a dumped .cia file and one downloaded online. Extracting romfs and placing it within patch folders still causes an error.

Tyulis, 2017-11-23:
Placing the entire romfs in the luma folder is not a good idea to avoid slowdowns. Then, do what i said before: go to sysnand sd/titles/0400000e/00175e00/content, mount the .app to drive and overwrite the dumped files with these ones. I already got this issue before, it's probably a conflict with the update.

BizarreDio, 2017-11-24:
I've done that too, I've dumped the .cia in every way possible, encrypted and decrypted. Tried it with GodMode9 and PackHack, still getting errors. Is there any video tutorial I can follow? The one I found uses SaltySD and I can't find out how to install that, not sure even if I need SaltySD anyways.

Tyulis, 2017-11-24:
SaltySD is old and bugged, it's the best way to get more errors. From where does your cia come? Actually, the only possibility i see is a problem with it.

BizarreDio, 2017-11-24:
Originally my CIA was from FreeShop, it's not a legitimate copy and its a digital download. The other CIA I use is from a ChaosGames thread, downloaded it and used that.

BizarreDio, 2017-11-24:
I'm positive the CIA I downloaded from ChaosGames was an actual legitimate copy, however.